markwal/GPX #3: Replicator 2x HBP max temp

ve3sjk, 2016-01-22:
I use this plugin as part of Simplify3d and the current version in my software has a max temp fix of 120C. I use a glass plate and need 130C to print anything. Is there any instructions i can get on how to compile it with a new max temp under windows using cgywin. I have cgywin and gcc installed but i don't know the procedure to compile it. The issue is on line 74 of gpx.h

#define HBP_MAX 120

dcnewman, 2016-01-22:
On 21/01/2016 8:39 PM, Steve wrote:
> I use this plugin as part of Simplify3d and the current version in my software has a max temp fix of 120C
>  I use a glass plate and need 130C to print anything Is there any instructions i can get on [...]

Use gnu make and issue the command "make" from a terminal window whilst in the top-level GPX
source directory.  You may need to install other build tools as well (e.g., gnu make, the
gcc tool chain, etc.).  I'm also not sure that GPX builds under cygwin.  I know that when
I need to build it under windows, I use mingw (www.mingw.org).

Can your build plate even achieve a control temp of 120C?  Many cannot -- they take so long
without ever reaching that temp that the firmware declares and error and disables the HBP.

However, it seems somewhat odd that you need a temp of 130C because you use a glass plate.
For instance, I've used glass plates for years on all my printers and I've never needed a
temp above 100C for everything I print (ABS, PLA, PC, HIPS, PVA, PETG, nylon, etc.).

Finally, note that the MakerBot-style firmwares impose a max of 130C for the HBP.  So,
don't bother attempting a value higher than 130C in GPX unless you are also prepared
to build the firmwares as well.  (And they don't build on Windows.)

Dan

dcnewman, 2016-01-22:
>  I'm also not sure that GPX builds under cygwin.  I know that when
> I need to build it under windows, I use mingw (www.mingw.org).

My mistake: I see that Mark says that it does build under cygwin (but using some tools
from mingw).  It'll also build for Windows using a Linux system and the mingw cross-compiler
tools on Linux.  (That's actually my preferred method for building things for Windows.)

Dan

markwal, 2016-01-22:
Perhaps we should switch it to 130 to match the Sailfish sources?  In GPX it is also used in macros to decide whether you mean the nozzle or the bed, but it doesn't seem like anybody would be relying on it being 120 for that purpose?

dcnewman, 2016-01-22:
On 21/01/2016 10:32 PM, Mark Walker wrote:
> Perhaps we should switch it to 130 to match the Sailfish sources?

... and makerbot, although they've hard coded the 130 in several places so it's
harder to spot.

Frankly, I didn't realize GPX had that hardcoded.  I'm not too sure why it's there
at all -- may date back to some historical code in RepG.  Personally, I'd remove
the limit other than to constrain it to the proper bit width.  Then leave it to the
firmware to assert what it understands to be the proper limit for the hardware.
OTOH, caution suggests just raising it to 130C.

Dan

markwal, 2016-03-29:
2.4.2 now does not limit the maximum bed temperature.  It's now up to your firmware and hardware to keep MakerBot and RepRap's disagreement over gcode from making a fire.

closed 2016-03-29 by markwal