markwal/GPX #4: Wrong handling of M190

CoDanny, 2016-02-10:
I just found an odd behavior with the M190 command. It seems as if when 'M190 S60' is the only command indicating bed temperature, GPX will not produce a x3g command to heat the bed.

It looks like M190 is interpreted as wait for bed temperature to reach a certain level, previously indicated, presumably by an M104. Slic3r however seems to only add a M190 with the temp at the beginning.

Would it be possible to change the behaviour of GPX so that it uses the temperature value stablished by the M190 command?

markwal, 2016-02-11:
My sources say M190 with S_IS_SET should call set_build_platform_temperature and then wait, but only if the current tool's (or the indicated one with Tn) thinks it has a heated build platform.

https://github.com/markwal/GPX/blob/master/src/gpx/gpx.c#L5450

I think there was a fix earlier to make it so that the machine profiles that think A has a build platform also think B has a build platform (for modern machines).  What machine profile and tool were you using when you reproduced the issue?

markwal, 2016-02-11:
Of course, the wait also only happens under those conditions.

dcnewman, 2016-02-11:
> I think there was a fix earlier to make it so that the machine profiles that think A has a build platform also think B

Correct.  I did that a while back.

CoDanny, 2016-02-11:
Thanks for this comments. It turns out that I was using the latest release from Dan's repo. Sorry for the confusion. I have now installed and tried 2.3 from this repo and everything works fine

closed 2016-02-11 by CoDanny