markwal/GPX #10: "Refresh SD files" stopped working.

giantpune, 2016-03-29:
My setup is OctoPi with GPX plugin connected to a Flashforge Creator Pro with a Toshiba Flashair sd card.  Previously, I was able to upload files to the Wireless SD card, then use "Refresh SD Files" in OctoPrint and find the new file in the file list to print it.  

I'm currently on OctoPrint: 1.2.10 and GPX Plugin: 2.4.2.  When I upload my new file, i cannot get that file to show up in the file list to print.  I do have quite a few existing files still showing up, though.  When i use the printer's buttons, I can manually find the file and print it.

The new file is the same extension as all the old files and in the same folder.  When I restarted octoprint, it refreshed its file list and was able to show my new file.

markwal, 2016-03-29:
What was the filename specifically?

When you click "Refresh SD Files", did the filename go by in the Terminal window?  (trying to figure out if it is the firmware not enumerating it, or the UI not showing it)

giantpune, 2016-03-29:
filename is paintbrush.x3g

markwal, 2016-03-29:
In OctoPrint, the "Terminal" tab shows the conversation between OctoPrint and GPX in gcode.

giantpune, 2016-03-30:
I just created another file called paintbrust.x3g and sent to the sd card.  This time when I clicked the refresh sd files, everything worked correctly.  Then new file shows up along side all the existing ones (I see the filename in the terminal tab as expected).

So i guess its a Heisenbug.  I'll let you know if I can find a way to reproduce it.

I did try clicking the delete button to delete the file, but the delete button does not work.  I see an error in the terminal.
```
Recv: End file list
Send: M105
Recv: ok T0:30 /0 T1:30 /0 B:21 /0
Send: M30 /paintbrust.x3g
Recv: ok
Recv: // echo: (line 17971) Syntax error: unrecognised gcode '/paintbrust.x3g'
Send: M20
Recv: ok
Recv: Begin file list
```

markwal, 2016-03-30:
Yeah, the x3g protocol doesn't have a command for "delete SD card file" so there's nothing to translate the M30 to.  I really ought to add some descriptive error messages into the plugin for stuff like that.  (Upload to SD card too, for example).

giantpune, 2016-04-16:
Ok, I've just reproduced the issue.  It happens after I print a file, the plugin doesn't realize that the print has stopped.  Clicking the "refresh sd card" button doesn't produce any output at all in the terminal tab.  It is still waiting for the job to complete, which happened hours ago.

I think my bug is caused by this one https://github.com/markwal/OctoPrint-GPX/issues/16 which is supposedly fixed.  I'm updating now.

markwal, 2016-05-12:
@giantpune are you still able to reproduce?

giantpune, 2016-05-12:
You can consider this fixed.  It stopped happening after I got https://github.com/markwal/OctoPrint-GPX/issues/16 fixed

closed 2016-05-12 by giantpune