markwal/GPX #15: Cygwin on Windows10: Compile erros for multiple definitions and undefined references.

btodoroff, 2016-05-08:
```
gcc  -g -O2   -o gpx.exe gpx.o gpx-main.o ../shared/machine_config.o ../shared/opt.o vector.o winsio.o -lm
winsio.o: In function `gpx_sio_open':
/home/Brian/GPX/build/src/gpx/../../../src/gpx/winsio.c:40: multiple definition of `gpx_sio_open'
gpx-main.o:/home/Brian/GPX/build/src/gpx/../../../src/gpx/gpx-main.c:208: first defined here
winsio.o: In function `gpx_sio_open':
/home/Brian/GPX/build/src/gpx/../../../src/gpx/winsio.c:47: undefined reference to `_open_osfhandle'
collect2: error: ld returned 1 exit status
Makefile:371: recipe for target 'gpx.exe' failed
```
Cloned from commit  051ee5c. Changed "#include <time.h>" to "#include <sys/time.h>" to fix unknown size of timeval in gpx.c.  Cygwin under Babun. GCC 4.9.2.  Windows10 64bit.

markwal, 2016-05-09:
This is true.  I've never intended it to actually build using cygwin directly.   I use cygwin only as an environment to run the mingw compiler.  I could add a better error message to make.

markwal, 2016-05-09:
If you install the mingw compiler under cygwin via setup.exe, you can use it this way:

mkdir build
cd build
../configure --host=x86_64-w64-mingw32
make
make install

closed 2018-06-19 by markwal