markwal/GPX #17: Random stops when using GPX to send gcode directly to printer

richyguitar, 2016-08-02:
Hi,

I'm very new to the whole matter so please let me know if I did something wrong or missed information.

I want to make my 3D printer somewhat intelligent by using the RasPi as a "control device". Therefore, I want to use command line only tools to talk with the printer.

I am using a Raspberry Pi 2 B with the OctoPrint image. I installed gpx via apt-get.
I had to install OctoPrint version 1.2.14 since the GPX-plugin does not support the current 1.2.15 version.

I try to print a 20mm testing cube from thingiverse (if I process it with MakerWare in my Windows machine and put it on SD it works perfectly fine).

When I try to send a gcode file to my Replicator 2X printer (Sailfish, latest firmware), the build process stopps at a (for me) random point. The syntax I use is: (also tried -s, did not change anything for me)
gpx -v -g -m r2x file.gcode /dev/ttyACM0
The printer is connected to USB (ttyACM0).
The output is looking fine, it loads the machine definition, reading the file and writing to /dev/ttyACM0. But the x3g output size is a somewhat random number (always a couple 100 bytes, but (almost) never the same size).

When I output to a .x3g file instead of ttyACM0 the filesize is always the same (104629 bytes, just fyi).

Am I doing something wrong with the command line?

Thanks for the help,
Richard

markwal, 2016-08-02:
GPX isn't intended to be a print host by itself. The serial output it's included to be able to have short scripts that change eeproms.

The plugin has been updated to support 1.2.15.

To print gcode with OctoPrint, upload the gcode using OctoPrint's web interface and click print. Then OctoPrint is the host and the plugin uses GPX under the covers line by line (no command line necessary).

It works for me on most things. However, printing over serial does not work nicely for smooth cylinders (anything that slices to a many segmented circle of say 30mm or larger) with speeds faster than about 40mm/s because the data rate exceeds the call response protocol and you get stutters. For those, use your printer's SD card or slow down.

closed 2016-08-02 by markwal

richyguitar, 2016-08-03:
Thank you. I have to control the printer with external software I have to write. Is there any chance to not use the OctoPrint webinterface for that but command line tools? I could not find any :(

markwal, 2016-08-03:
You could try the Makerbot s3g python library, if you wanted to control it from python: https://github.com/makerbot/s3g

You might also be able to do it using octoprint's command line interface (devel branch).

Or write your own command line controller that talks to octoprint via REST.

markwal, 2016-08-03:
Google also led me to this: http://octocmd.readthedocs.io/en/latest/

richyguitar, 2016-08-03:
Thank you again very much, apreciate it!
I gave octocmd a try but did not work as planned, might be my fault tho.