markwal/GPX #18: Output percentage progress when streaming a file

matthova, 2017-02-21:
Curious how easy it would be to output percentage progress, either with a flag or with the verbose flag. Or is there an easier alternative to track the progress of a file being streamed?

markwal, 2017-02-26:
Actually gpx wasn't intended to be a gcode sender and it lacks many features of such a thing including displaying progress, but also pause, cancel, terminal input, temperature indication, etc.  It's usually used offline to convert a file to x3g format.  The serial mode is usually used to send short messages from the command line to update the eeprom and the like or is orchestrated by another program like OctoPrint.

markwal, 2018-06-19:
While this request is still outstanding, I'd like to close since there aren't any current plans to move the sender portion of GPX along.  The translated x3g can contain progress (GPX has a flag) and several senders also emit progress.

closed 2018-06-19 by markwal