markwal/GPX #21: {ERROR} M116 Doesn't seem to be working??

RJ-Make, 2017-03-15:
Not sure if this is gpx bug, or an OctoPrint Bug.

I'm running OctoPrint version 1.3.2rc1
Here is my starting G-Code, which doesn't seem to be working now, specifically the M116 Command

Observations:
1) On a cold start, both bed and extruder temps show up on the temperature tab.
2) As soon as the bed temperature (60C Setpoint) reaches temperature the print starts, with only a slight increase in extruder temp.
3) On a warm bed start (say 10C below actual temp), the extruder temp is not passed to OctoPrint, and the print again starts. 
4) I've notice that is takes 15-30 seconds on a cold start for the extruder set point temp to show up in OctoPrint

If I replace:

> M140 S[bed0_temperature] T0 ; Heat buildplate
M104 S[extruder0_temperature] T1 ; Heat left extruder
M116 ; Wait for all temps to be reached

With this, everything works fine. 

> M140 S[bed0_temperature] T0
M134 T0 ; stabilize bed temperature
M104 S[extruder1_temperature] T1
M133 T1 ; stabilize left extruder temperature

No Longer Works
> ; **** Replicator 1 dual start.gcode Left ****
T0; always home on the right extruder
M73 P0 ; Enable build progress
G162 X Y F3000 ; Home XY maximum
G161 Z F1200 ; Home Z minimum
G92 X0 Y0 Z-5 A0 B0 ; Set Z to -5
G1 Z0 ; Move Z to 0
G161 Z F100 ; Home Z slowly
M132 X Y Z ; Recall home offsets
G1 X-95 Y-70 Z30 F9000 ; Move to wait position off table
T1 ; switch to the left extruder
M126 S[fan_speed_pwm] ; Set fan speed
M140 S[bed0_temperature] T0 ; Heat buildplate
M104 S[extruder0_temperature] T1 ; Heat left extruder
M116 ; Wait for all temps to be reached
G130 X118 Y118 Z40 A118 B118 ; Default stepper Vrefs
G92 A0 B0 ; Zero extruders
G1 X100 Y-70 F9000 ; Move to front right corner of bed
G1 Z0.3 F6000 ; Move down to purge
G1 X-90 Y-70 E24 F2000 ; Extrude a line of filament across the front edge of the bed
G1 X-100 Y-70 F180 ; Wait for ooze
G1 X-110 Y-70 F5000 ; Fast wipe
G1 Z0.5 F100 ; Lift
G92 A0 B0
M73 P1 ; @body (notify GPX body has started)
; **** end of start.gcode ****

markwal, 2017-03-16:
Hmmm...  I don't hit this issue maybe because I do my M140 and M104 right after M73 P0 (may as well be heating up while homing).

However, in your case, if that's a copy and paste and not a typo, the reason it works differently isn't the M116 (which emits the same x3g as M133 anyway), it is the parameter is different:

Everything works fine:
M104 S[extruder1_temperature] T1

No longer works:
M104 S[extruder0_temperature] T1

Maybe it is the extruder0?

But my order looks like this:

M73 P0
M140
M104
T0
G130 X118 Y118 A118 B118
G162
...

That way you tell the bot and the plugin you're printing as soon as possible, set the temps, switch extruder, assert the right power on the steppers so that homing works right, then home.

RJ-Make, 2017-03-16:
Thanks Mark, I'll give those a shot tonight..

RJ-Make, 2017-03-16:
Mark would you mind posing you entire start up script? I don't see M116

markwal, 2017-03-16:
I don't have Simplify3d so my expansion variables are different and I've ditched the bead across the front and made a few other tweaks since I posted them: http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html

RJ-Make, 2017-03-16:
Thanks Mark

markwal, 2018-06-19:
My guess is that this is no longer an issue.  If I'm wrong, feel free to comment and I'll reopen.

closed 2018-06-19 by markwal