markwal/GPX #25: CTC printer (replicator one clone settings)

kyuzumaki, 2017-10-03:
The scaling by default does not seem to be correct with the cr1d profile. When I used the default setting my parts which should be 27mm across came out at 25.8mm. After some research and editing I have made this config file which works 1:1 scale for the printer. Essentially steps_per_mm = 94.40389

Could this be updated in the main branch? I see its set to 88.8 in std_machines.h

dnewman-polar3d, 2017-10-03:
Sounds like you received a Clone R1 with 17T GT2 timing pulleys instead of 18T GT2 timing pulleys.  Did you change your timing pulleys or did you ask Carl for one like that?

Before this is changed, this needs to be confirmed with Carl: did he ship all Clone R1's with 18T or 17T timing pulleys?

kyuzumaki, 2017-10-03:
Oh I think I've misunderstood the naming. I thought the r1d clone was generic meaning the common R1D clones i.e. CTC printers. In any case this could be used for a CTC specific config, I did a few rough measurements myself which were close to what I found on this page where someone has taken more detailed measurements
http://www.dragonflydiy.com/2015/07/ctc-printer-tuning.html

stuartpb, 2017-12-30:
Yeah, what "clone" does this refer to? It seems like the clone profile also has a higher nominal packing density, which sounds like it has something to do with the presence of the heated build plate, which is true for R1D clones like the CTC, Flashforge Creator Pro, and Qidi Tech I.

dnewman-polar3d, 2017-12-30:
The "clone" reference in the original post is NOT a Replicator 1 clone.  It is a Clone R1 by Carl Raffle.  The original poster incorrectly assumed it meant some flavor of a Replicator 1 clone.  A CR1 is most definitely not a Replicator 1 clone given that it uses a Core-XY drive train.

As to packing density, there is no one packing density/extrusion factor that fits all Replicator 1 clones.  It's across the board.   Just one of several reasons why you should always calibrate your slicer to your printer for each type of filament you use.  E.g., printing 100% filled 20 x 20 x 10mm calibration boxes.  Carefully measure the filament diameter before slicing and input that to the slicer.  Then print. Then determine if the top is concave, convex, or really nice and flat.  If not flat, then adjust the packing density/extrusion factor, reslice, and reprint.

stuartpb, 2017-12-30:
Ah, okay. From https://www.reddit.com/r/3Dprinting/comments/2hkimt/clone_r1_corexy_printer_needs_you/:

> The concept started life as a Replicator clone, and is named the Clone R1, but along the way it changed into the far-superior CoreXY gantry.

I might submit a PR to rename the profile accordingly, because it really is confusing as currently named (it sounds like it refers to any Replicator clone with a heated build plate).

As for the packing density, that sounds like something that's controlled by the parameters that are passed into the slicer, no? Why does GPX have a setting for it at all, since its role is (as I understand it) to translate G-code, as defined by the slicer according to its own packing density settings, into x3g commands? Shouldn't anything that needs to know about packing density have already done the math before GPX sees it?

dnewman-polar3d, 2017-12-30:
GPX has an input for it for when you use  macros that allow, mid print, switching to a different filament with it's own diameter and packing density.  GPX will then make both volumetric adjustments as well as compensate for different packing densities (as it can vary between filament types).  But to do those adjustments it needs to know the original filament diameter and packing density for which the gcode was generated.

markwal, 2018-06-19:
I think the original issue was that the wrong printer profile was selected for the CTC and that's been resolved, right?  If not, feel free to comment and I'll re-open.

closed 2018-06-19 by markwal