markwal/GPX #26: fcp machine type missing

jmacfarland, 2017-11-16:
Installed latest on ubuntu 16.04, using apt install gpx.

when I run gpx, it shows the list of all commands/machine types/etc, but fcp is not on the list of machines.

jmacfarland, 2017-11-16:
Note: fcp is still present in GpxUi.

dnewman-polar3d, 2017-11-16:
Fcp is just an alias for r1d (Replicator 1 Dual).  Select r1d.

jmacfarland, 2017-11-16:
Ah okay, thank you!

markwal, 2017-11-17:
Also, the alias functionality of gpx is new for later versions.  I'm not sure what version you get through apt install at the moment.  The deb package is maintained by @hyperair.

closed 2017-11-17 by markwal

hyperair, 2017-11-17:
2.5.2 has been available for a while. https://launchpad.net/ubuntu/+source/gpx

hyperair, 2017-11-17:
Hmm, I think xenial may not have it. You can get 2.5.2 for Ubuntu 16.04 from https://launchpad.net/~hyperair/+archive/ubuntu/3dprinter/+packages:

```sh
sudo add-apt-repository ppa:hyperair/3dprinter
sudo apt update
sudo apt install gpx
```

jmacfarland, 2017-11-17:
That did it, thanks!