markwal/GPX #32: M135 command

YHare, 2018-06-12:
MakerBot Desktop using M135 command to change to different tool. (M135 T0, M135 T1).
Unfortunately GPX converter ignore that  commands. 
It's make very difficult using OctoPrint with GCode produced by MakerBot Desktop.  
On the other hand GPX converter translates simplest commands  T0,T1 to change to different tool. 
How difficult will be to add M135 command to GPX converter.

markwal, 2018-06-19:
GPX handles M135 as a tool change.  See: https://github.com/markwal/GPX/blob/master/src/gpx/gpx.c#L5695

markwal, 2018-06-19:
BTW There are versions of RepG that emit M108 for toolchange and that got propogated into Slic3r.  Earlier versions of GPX ignored that, but recent versions handle that also.

closed 2018-06-19 by markwal