markwal/GPX #40: Heatsink fan for Makerbot TOM

michaelbaisch, 2020-08-02:
Hello,

I'm now running into the problem that I can't seem to be able to turn the heatsink fan of my Makerbot TOM on. I assume [`set_Fan`](https://github.com/markwal/GPX/blob/f22b938e773f4accf9138d8723a4ecfbdefdd2b2/src/gpx/gpx.c#L1530) would turn it on, but when the machine type / flavor is set to MakerBot, it looks like there is no way to enter this function. `M016` only calls `set_fan`, `if(gpx->machine.id >= MACHINE_TYPE_REPLICATOR_1)`.

With ReplicatorG I used `M106` to turn on the fan, so I'm not sure why this is different here and the Automated Build Platform is enabled. It has been a long time since I used the ABP but I'm pretty sure that there was also a way to also turn the fan on. So I guess my question is if that check for `MACHINE_TYPE_REPLICATOR_1` could be removed? And I believe that everyone still working with a TOM has dismantled the ABP since the results are not that good.

Greetings,
Michael

markwal, 2020-08-26:
Seems legit. I haven't had a TOM myself and that code is from wingcommander.

markwal, 2020-08-29:
So should we come up with a different M-code for controlling the ABP or just assume that no one uses the ABP any more?  Or at least no one who is using GPX.

markwal, 2020-08-29:
We could make a different machine type (TOM+ABP) and make that the one that gets the ABP codes.  It kinda looks like that's how RepG handled it: https://github.com/makerbot/ReplicatorG/blob/d6f2b07785a5a5f1e172fb87cb4303b17c575d5d/machines/thingomatic/end%2BABP.gcode

tinder-danielnewman, 2020-08-30:
A new machine type makes the most sense.  Gosh, I still have an ABP in a box somewhere.   Just didn't work too well.  People actually sold thin sheets of titanium you could use to make a belt out of for the ABP.  Seemed a bit excessive to me.

markwal, 2020-08-30:
First take: 416ef0cb829900d82f01f41cb817868bc063adf9

michaelbaisch, 2020-08-30:
Thanks for your work! 

I can test it in a few days when I have access to my printer again. About lines 5314 and 5357 `if(gpx->flag.reprapFlavor || gpx->flag.M106AlwaysValve) {` if I recall correctly, I believe `M106AlwaysValve` was set true for me, at least I thought that this if statement was entered. If that could be right, is there a way to set this from OctoPrint-GPX?

(I just modified this line to `if(gpx->machine.id >= MACHINE_TYPE_REPLICATOR_1 && (gpx->flag.reprapFlavor || gpx->flag.M106AlwaysValve)) {`)

michaelbaisch, 2020-09-06:
I tested your changes and as I suspected there was something with `if(gpx->flag.reprapFlavor || gpx->flag.M106AlwaysValve) {` in fact both flags are true for me. Not sure how to set them correctly especially, `flag.M106AlwaysValve` and from OctoPrint-GPX. OctoPrint-GPX prints this info:

```
octoprint.plugins.GPX - INFO - {'a': {}, 'None': {}, 'b': {}, 'machine': {}, 'printer': {'recalculate_5d': '0', 'gcode_flavor': 'makerbot', 'build_progress': '1', 'machine_type': 't7', 'ditto_printing': '0'}, 'right': {}, 'y': {}, 'x': {}, 'z': {}, 'left': {}}
```

markwal, 2020-09-07:
Hang on.  The M106 command that we're trying to interpret here is to turn on the print cooling fan, not the heatsink fan.  IE on my FlashForge it's the Extra Fan pins from the motherboard not the heatsink fan.

The sailfish I'm running always turns on the heatsink fan whenever the nozzle temperature exceeds 50C, so I don't do any g-code specifically for the heatsink fan.

I'm gathering the TOM doesn't do that?  You get to pick whether to do heatsink or ABP?  That doesn't seem right.  It should be print cooling fan or ABP, right?  And Sailfish always turns on the heatsink fan?

dcnewman, 2020-09-08:
Heat sink fan for ToMs (as well as other Makerbots) has no Gcode command.  It's controlled automatically by the firmware.  Only real differrence between the G3 firmware (Cupcake, ToM) and mightyboard fw is that you could set the on/off temp for the G4 firmware via EEPROM and RepG.  Don't believe you could do that for the mightyboard fw.

The ABP is controlled [here in Motherboard.cc](https://github.com/jetty840/Sailfish-G3Firmware/blob/master/firmware/src/Motherboard/Command.cc#L1591).   That s3g command is [here](https://github.com/jetty840/Sailfish-G3Firmware/blob/master/firmware/src/shared/Commands.hh#L154).  And some of the ReplicatorG code is [here](https://github.com/jetty840/ReplicatorG/blob/master/src/replicatorg/drivers/gen3/Sanguino3GDriver.java#L1738-L1762) and [here](https://github.com/jetty840/ReplicatorG/blob/master/src/replicatorg/drivers/commands/ToggleAutomatedBuildPlatform.java).

michaelbaisch, 2020-09-08:
Thanks @dcnewman! I booted up an older machine with the current sailfish repg and indeed I found a disabled option there “Enable regulated cooling fan”.

[image]

Never knew this was there. For me M106 always did the job of turning the heatsink fan on. And with the previous discussed changes it worked again from the Octoprint Terminal or the Control panel.

So this kind of fixes the problem for me. Although some things still seem to be a bit contradictory.

markwal, 2020-11-18:
OK. So I propose we close this for now and revisit if we get a new request for print cooling fan control instead of ABP.

closed 2020-11-18 by markwal