markwal/GPX #42: gpx 2.6.7 build failure

chenrui333, 2020-11-21:
👋 trying to upgrade gpx to the latest release, but running into some build issue.

Here is the build failure log

<details>
<summary>build failure</summary>

```
==> make install
Making install in src/gpx
clang -DHAVE_CONFIG_H -I. -I../../src/shared  -Wall -Wstrict-prototypes -Wformat -Werror=format-security -DSERIAL_SUPPORT -I../../src/shared   -g -O2 -c -o gpx.o gpx.c
clang -DHAVE_CONFIG_H -I. -I../../src/shared  -Wall -Wstrict-prototypes -Wformat -Werror=format-security -DSERIAL_SUPPORT -I../../src/shared   -g -O2 -c -o gpx-main.o gpx-main.c
clang -DHAVE_CONFIG_H -I. -I../../src/shared  -Wall -Wstrict-prototypes -Wformat -Werror=format-security -DSERIAL_SUPPORT -I../../src/shared   -g -O2 -c -o gpxresp.o gpxresp.c
clang -DHAVE_CONFIG_H -I. -I../../src/shared  -Wall -Wstrict-prototypes -Wformat -Werror=format-security -DSERIAL_SUPPORT -I../../src/shared   -g -O2 -c -o vector.o vector.c
clang -DHAVE_CONFIG_H -I. -I../../src/shared  -Wall -Wstrict-prototypes -Wformat -Werror=format-security -DSERIAL_SUPPORT -I../../src/shared   -g -O2 -c -o ../shared/machine_config.o ../shared/machine_config.c
clang -DHAVE_CONFIG_H -I. -I../../src/shared  -Wall -Wstrict-prototypes -Wformat -Werror=format-security -DSERIAL_SUPPORT -I../../src/shared   -g -O2 -c -o ../shared/opt.o ../shared/opt.c
clang  -g -O2   -o gpx gpx.o gpx-main.o gpxresp.o ../shared/machine_config.o ../shared/opt.o vector.o  -lm 
Undefined symbols for architecture x86_64:
  "_iconv", referenced from:
      _translate_handler in gpxresp.o
  "_iconv_close", referenced from:
      _translate_handler in gpxresp.o
  "_iconv_open", referenced from:
      _translate_handler in gpxresp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [gpx] Error 1
make: *** [install-recursive] Error 1
```

</details>

full build log is in here, https://github.com/Homebrew/homebrew-core/runs/1424779985
relates to https://github.com/Homebrew/homebrew-core/pull/65214

chenrui333, 2020-11-26:
cc @markwal

markwal, 2020-11-28:
@chenrui333 Thanks for bringing this to my attention.  I didn't add the iconv dependency correctly for configure to handle it.  I've fixed in the latest master.  Now I need to make a 2.6.8 release...

closed 2020-11-30 by markwal