markwal/GPX #43: Dual extrusion - left extruder wont warm up

Meganium97, 2021-06-25:
I'm using Prusaslicer and post processing with gpx, and I set prusaslicer up with all the things it needs, but when I go to print dual, the file that I've set to use extruder "2" won't print, and the temperature never even updates on the 3d printer.
Here is my start gcode
```
T0; set primary extruder
M73 P0; enable show build progress
M109 S[first_layer_bed_temperature] T0; heat bed up to first layer temperature
M104 S[first_layer_temperature_0] T0; set nozzle heater to first layer temperature
M103; disable RPM
G21; set units to mm
G162 X Y F6000; home XY axes maximum
G161 Z F9000; home Z axis minimum
G92 Z0; set Z to 0
M132 X Y Z A B; Recall stored home offsets
G90; set positioning to absolute
G1 Z30; move Z to waiting height
G1 X-95 Y-73 Z30 F14000; move to waiting position (front left corner of print bed)
G130 X0 Y0 A0 B0; set stepper motor vref to lower value while heating
M6 T0; wait for bed and extruder to heat up
G130 X127 Y127 A127 B127; set stepper motor vref to defaults
M108 T0 R3; set extruder speed
G92 E0; set E to 0
G90; use absolute coordinates
M320; acceleration enabled for all commands that follow
G1 Z[first_layer_height] F6000.000; move to first layer height
G1 X100 Y-73 F14000.000; move to front right corner of bed
G1 X-90 Y-73 E24 F2000.000; extrude a line of filament across the front edge of the bed
G4 P2000; wait for ooze to slow
G1 Z0 F6000.000; lower nozzle height to 0
G1 X-95; wipe nozzle
G1 Z[first_layer_height] F6000.000; set nozzle to first layer height
G1 F12000; ensure fast travel to first print move
G92 E0; set E to 0 again
M73 P0; reset build progress to 0
```

And here is my end gcode:
```
M73 P100 ; end build progress
G162 X Y F2500; home X and Y axes
M18; disable all stepper motors
M109 S0 T0; set bed temperature to 0
M104 S0 T0; set extruder temperature to 0
M70 P5; Display "We <3 Making Things!" message
M72 P1; Play Ta-Da song
```
Am I doing anything wrong?

CTcartesian3D, 2021-06-25:
The left extruder is = T1
The right extruder is = T0

Meganium97, 2021-06-25:
Thanks. I now realized that I never actually made it check for the temp of the second nozzle. I put that in and now it works flawlessly.

closed 2021-06-25 by Meganium97