markwal/GPX #46: Unsupported Printers - Advice in ReadMe? FAQ?

jbarber877, 2022-08-13:
Hi, 
I'm wondering how we should deal with unsupported printers - should we just leave default settings? Or should we try to guess what model is closest to what we are using? Some advice in the readme, or perhaps an FAQ page somewhere would be helpful.

On a personal note, I am tyring to get an older model Klic-N-Print (a gift from a friend who upgraded to a newer printer) up and running again. The company is out of business, and I have discovered that most of the software intended to be used with the printer is no longer available for download. The only reason I bother with this printer is the conveinence of dual extruders. If anyone knows what settings to use with this brand, some info would be greatly appreciated. Otherwise, I will experiment with the software and maybe post a reply if I find something that works.

smerkousdavid, 2022-09-25:
Hey I was able to find the Klic-N-Print converter.exe. I then used IDA to view the exe and found they're just calling GPX in the backend with the machine definition of r1d and makerbot/replicator GCODE flavor, and override build perc.

[image]

To get the profiles/KNP3D converter look at my comment here. I've been trying to piece things together for this unsupported printer. https://www.reddit.com/r/3Dprinting/comments/kcp9dx/klicknprint_knp3d_converter/

jbarber877, 2022-09-25:
Thanks, I'll check it out!

smerkousdavid, 2022-09-30:
> Thanks, I'll check it out!

Any luck? I've tried it with OCTOPRINT GPX plugin and it works but for some odd reason I get a lot of layer skips. I can generate GPX and x3g file directly to SD and it's fine, but the same print through octoprint has issues. So I'm guessing it's either octo/octo-gpx but wanted to confirm with you.